LCAS/spqrel_tools #11: GPSR

marc-hanheide, 2017-07-13:


iocchi, 2017-07-19:
Roberto provides examples of GPSR commands.
Andrea provides examples of LU4R output.
Roberto generates plans from LU4R output.

robpinillos, 2017-07-21:
Some examples generated by the last version of GPSRCmdGen:

CATEGORY 1
##########

Take the noodles from the dining table to the bar
Go to the office, locate a person, and answer a question
	question
		Q: What time is it?
		A: ...
Look for the soap in the kitchen
Tell me how many tea there are on the bar
Give to Skyler at the desk the coke from the counter
Look for the tea in the bathroom
Pick up the pringles from the bedside and put it on the sideshelf
Answer a question to Michael at the sink
	question
		Q: What is the highest point in Japan
		A: The highest point in Japan is Mount Fuji, which reaches 3776m above sea level
Go to the corridor, look for a person, and say your team's affiliation
Find the beer in the kitchen
Take the peanuts from the living table and deliver it to Hanna at the bed
Put the tea on the desk
Put the noodles on the living shelf
Find the pringles in the corridor
Say a joke to Hanna at the dining table

CATEGORY 2
##########
Navigate to the cupboard, look for the mints, and deliver it to me
Navigate to the bedroom, look for a waving person, and answer a question
	question
		Q: Who invented the first compiler
		A: Grace Brewster Murray Hopper invented it
Follow Sophie from the TV stand to the bedroom
Take Sophie from the dining table to the cabinet
	remarks
		 The person being guided must deviate when indicated by the referee before reaching the cabinet
Tell me how many mints there are on the bar
Tell me how many paprika there are on the bookcase
Meet Erika at the desk and escort her to the bed
	remarks
		 The person being guided must deviate when indicated by the referee before reaching the bed
Tell me how many people in the corridor are boysTell me how many people in the corridor are boys
Accompany Samantha from the sink to the cabinet
	remarks
		 The person being guided must deviate when indicated by the referee before reaching the cabinet
Meet Ken at the desk and take him to the sink
	remarks
		 The person being guided must deviate when indicated by the referee before reaching the sink

andreavanzo, 2017-07-21:
CATEGORY 1
##########

Take the noodles from the dining table to the bar|TAKING(theme:"the noodles"source:"from the dining table to the bar")
Go to the office locate a person and answer a question|Go to the office locate a person and answer a question

question
Q: What time is it?
A: ...

Look for the soap in the kitchen|LOCATING(phenomenon:"for the soap",ground:"in the kitchen")
Tell me how many tea there are on the bar|BEING_LOCATED(location:"on the bar")
Give to Skyler at the desk the coke from the counter|GIVING(theme:"to Skyler at the desk the coke",source:"from the counter")
Look for the tea in the bathroom|LOCATING(phenomenon:"for the tea",ground:"in the bathroom")
Pick up the pringles from the bedside and put it on the sideshelf|TAKING(theme:"the pringles",source:"from the bedside")#PLACING(theme:"it",goal:"on the sideshelf")
Answer a question to Michael at the sink|NO FRAME(S) FOUND

question
Q: What is the highest point in Japan
A: The highest point in Japan is Mount Fuji which reaches 3776m above sea level

Go to the corridor look for a person and say your team's affiliation|MOTION(goal:"to the corridor look for a person")#LOCATING(ground:"to the corridor",phenomenon:"for a person")
Find the beer in the kitchen|LOCATING(phenomenon:"the beer",ground:"in the kitchen")
Take the peanuts from the living table and deliver it to Hanna at the bed|TAKING(theme:"the peanuts",source:"from the living table")#BRINGING(theme:"it",beneficiary:"to Hanna at the bed")
Put the tea on the desk|PLACING(theme:"the tea",goal:"on the desk")
Put the noodles on the living shelf|PLACING(theme:"the noodles",goal:"on the living shelf")
Find the pringles in the corridor|LOCATING(phenomenon:"the pringles",ground:"in the corridor")
Say a joke to Hanna at the dining table|NO FRAME(S) FOUND

CATEGORY 2
##########
Navigate to the cupboard look for the mints and deliver it to me|LOCATING(ground:"to the cupboard",phenomenon:"for the mints")#BRINGING(theme:"it",beneficiary:"to me")
Navigate to the bedroom look for a waving person and answer a question|LOCATING(ground:"to the bedroom",phenomenon:"for a waving person")

question
Q: Who invented the first compiler
A: Grace Brewster Murray Hopper invented it

Follow Sophie from the TV stand to the bedroom|COTHEME(cotheme:"Sophie",source:"from the TV stand to the bedroom")
Take Sophie from the dining table to the cabinet|TAKING(theme:"Sophie",source:"from the dining table to the cabinet")
Tell me how many mints there are on the bar|BEING_LOCATED(location:"on the bar")
Tell me how many paprika there are on the bookcase|BEING_LOCATED(location:"on the bookcase")
Meet Erika at the desk and escort her to the bed|NO FRAME(S) FOUND
Tell me how many people in the corridor are boys|NO FRAME(S) FOUND
Accompany Samantha from the sink to the cabinet|COTHEME(cotheme:"Samantha",goal:"from the sink to the cabinet")
Meet Ken at the desk and take him to the sink|TAKING(theme:"him",goal:"to the sink")

iocchi, 2017-07-21:
Plan GPSR has been revised and action execplan completed and succesfully tested.
As you can see it is very simple.

execplan waits for a string for LU4R, it translates into a plan (TODO), then compiles this plan into a PNP and executes it.

The only missing part is the implementation of the function `LU4R_to_plan` that must
create a plan from the LU4R string.

closed 2017-08-04 by marc-hanheide