LCAS/spqrel_tools #13: SPR

marc-hanheide, 2017-07-13:


andreavanzo, 2017-07-20:
I've implemented most of the interactions of the task. We need to agreed about which information (and the sharing format) we can get from the vision module (e.g., size of the crowd, number of males and females, sitting people...) when analysing the crowd.

marc-hanheide, 2017-07-21:
I think we might just need an action `analyzecrowd`, that just looks around and try to get as much information from the people and populate a JSON data structure:
```
{
  `people`: [
    'id': 12342134,
    'gender': 'male'
  ]
}

marc-hanheide, 2017-07-21:
`TopologicalNav/CurrentNode`
`TopologicalNav/ClosestNode`

They raise events when changed

robpinillos, 2017-07-21:
There was incoherence between name of memory in the actions `memorypeople` and `memoryface`
Finally
`Actions/MemorizePeople/PeopleList/`
 `Actions/MemorizePeople/Person/<IDNUMBER>`

Btw Do you know how to delete all the data from:  `Actions/MemorizePeople/Person/<IDNUMBER#1>`,`Actions/MemorizePeople/Person/<IDNUMBER#2>`,
`Actions/MemorizePeople/Person/<IDNUMBER#3>`  prevously created?

robpinillos, 2017-07-21:
A little reminder about the info from each person. the info provided by Microsoft API  `age` and `gender` would be more reliable  (`face_ms_api` ) than the naoqi driver (`face_naoqi`). But the `face_ms_api` is only available after face learning ( It needs a face close to the robot)

closed 2017-08-04 by marc-hanheide