LCAS/spqrel_tools #19: Poster Session Poster

marc-hanheide, 2017-07-13:


marc-hanheide, 2017-07-13:
looking for ideas to what to include: What are the main scientific contributions?
* clearly nav (reference to paper) und speech

Please post ideas @iocchi @mtlazaro @sergimolina @pet1330 @robpinillos

marc-hanheide, 2017-07-13:
We may reuse this poster for the symposium, needs a good title

iocchi, 2017-07-16:
filled Planning & Execuion section. PNP of RIPS plan is included. If too difficult to read, we can put the plan and the ER instead.

mtlazaro, 2017-07-18:
Filled Navigation.
We should put photo of the Pepper maybe in the middle of all blocks, I've just left a picture there.

sergimolina, 2017-07-18:
Unfortunately, there are no novel contributions for the people perception since we are using Pepper API in the end :(

closed 2017-07-22 by marc-hanheide