LCAS/spqrel_tools #21: tablet actions

marc-hanheide, 2017-07-13:
use cases for tablet:

- start/stop task
- display speech output
- a general debugging display area for the tasks
- possibly: display answer options, ideally as a button, needs to interact with slu4r @andreavanzo

marc-hanheide, 2017-07-18:
some progress, the backend is working, needs integration with actions and some actual useful content. @andreavanzo is there a memory entry in `ALmemory` that contains the last utterance the robot said, so I can use display it on the screen?

andreavanzo, 2017-07-18:
sure! the entry key is named "Veply" (yes, it is supposed to be Reply, but the V is for Vanzo :D ). you just need to subscribe to that event

marc-hanheide, 2017-07-18:
Cool, I'll display that on the screen!

Another question @andreavanzo: 

Can i also "inject" a speech recognition into the dialog? My plan is to provide a speech_fake action that displays answers on the screen to shortcut actually saying them. In case speech recognition doesn't work as an implementation of the RoboCup CONTINUE rule.

andreavanzo, 2017-07-18:
But how do you choose the possible candidate answers?

marc-hanheide, 2017-07-18:
I was about to ask you 😉. The simplest solution would be a `offeranswerui` action with parameters for the options to be put in the plan before a dialog action. But I'm open to better ideas.

marc-hanheide, 2017-07-19:
Any suggestions? Or i implement my simple one.

marc-hanheide, 2017-07-19:
@andreavanzo @iocchi could you let me know how to inject the speech?

andreavanzo, 2017-07-19:
Hi @marc-hanheide sorry for the delay. I think the best solution is just to raise a new event, say "TabletAnswer" for the key and the selected answer as value. The dialogue module will subscribe to such event and handle the sentence properly.  This also let us to distinguish among sentences spoken by the user and those input from the tablet. Let me know if you agree

marc-hanheide, 2017-07-20:
Sounds good to me. I shall do that and then get ready.

marc-hanheide, 2017-07-21:
I'll make it listen also for `AnswerOptions` to display options coming from dialog.

marc-hanheide, 2017-07-21:
I have implemented both `AnswerOptions` and `TabletAnswer` as discussed above in 39a9543fc94abd7d3d877abf2670ccb75883578f

@andreavanzo The UI will react to a string of the format `"speechbtn_yes_no%it is not_third%option"` raised in `AnswerOptions` and render this as 

[image]

When one of the buttons is clicked, the text displayed is raised as an event on `TabletAnswer`.

marc-hanheide, 2017-07-21:
I have now also implemented the action `speechbtn`. Tested with 

`./action_cmd.py -a speechbtn -p "option_yes_no%it is not_third%option" -c start`

will display the three buttons as in [previous comment](https://github.com/LCAS/spqrel_tools/issues/21#issuecomment-317070425)

`./action_cmd.py -a speechbtn  -c start`

will remove them.

The action always returns successfully, immediately. It can hence, be included into any plan without any bigger side effects.

The action is implemented in fea8414ceea3df6a281a0b5509e56b7310976480

marc-hanheide, 2017-07-21:
tidied this all a bit more and used the "usual" action scheme. Looks final now.

iocchi, 2017-07-21:
@marc-hanheide  Need to reopen this issue because `#` and ` ` are not allowed in PNP. It works with action_cmd, but it won't work when the plan is translated in PNP. You can use `+` instead.
Example:  `speechbtn_yes_no+it+is+not_third+option`

marc-hanheide, 2017-07-21:
I used `%` and not `#`. I tried it and it worked fine. So, is that ok?

marc-hanheide, 2017-07-21:
Yes, whole tool chain, blockly->plan->pnml>naoqi_pnp. Worked just fine.

iocchi, 2017-07-21:
I get something strange with `%` that I do not understand (the plan seems to be correctly executed, but pnp_naoqi complains about a PNP condition with these symbols. While with `+` this does not happen.
I am not sure I'll have time to look into it. I suggest to use `+` I trust this situation better...

marc-hanheide, 2017-07-21:
OK, we shall move to `+` then.

commit d3244719 (2017-07-21): changed separator from `%` to `+` 

closes #21
closed 2017-07-21 by marc-hanheide