LCAS/spqrel_tools #22: speech recognition crashes robot

marc-hanheide, 2017-07-13:
@iocchi has seen this a couple of times.

andreavanzo, 2017-07-13:
I'd need some additional information on how to reproduce the crash or, at least, the log

iocchi, 2017-07-13:
This is a screenshot I could get when the problem happened

```
[W] 1499714730.460398 5887 qitype.dynamicobject: post on method failed:     ALPythonModule::execute
        calling SpeechRecognition_inst.text_done_callback
<type 'exceptions.RuntimeError'>
        ALAudioRecorder::stopMicrophonesRecording
        Socket Disconnected
[SpeechRecognition] Reset recording..
Traceback (most recent call last):
  File "speech_to_text/speech_recognition.py", line 193, in <module>
    main()
  File "speech_to_text/speech_recognition.py", line 189, in main
    sr.start()
  File "speech_to_text/speech_recognition.py", line 69, in start
    self._spin()
  File "speech_to_text/speech_recognition.py", line 148, in _spin
    self.reset()
  File "speech_to_text/speech_recognition.py", line 137, in reset
    self.audio_recorder.stopMicrophonesRecording()
  File "/opt/aldebaran/lib/python2.7/site-packages/naoqi.py", line 194, in __call__
    return self.__wrapped__.method_missing(self.__method__, *args, **kwargs)
  File "/opt/aldebaran/lib/python2.7/site-packages/naoqi.py", line 264, in method_missing
    raise e
RuntimeError:   ALAudioRecorder::stopMicrophonesRecording
        module destroyed
```

andreavanzo, 2017-07-19:
It should be solved, but we need some tests to reproduce the behaviour

closed 2017-07-20 by andreavanzo