LCAS/spqrel_tools #7: config file

iocchi, 2017-07-13:
Define a way to initialize parameters in config file(s)

Example:
read config file with <keys,values>
set all the keys with the values

marc-hanheide, 2017-07-13:
Should we make this one YAML file? 

Should also be used SPR, to define the answers to questions

Jailander, 2017-07-19:
I have made a script for this is part of this PR: https://github.com/LCAS/spqrel_tools/pull/35

Jailander, 2017-07-19:
Please check and close issue if happy

marc-hanheide, 2017-07-20:
Hi, do you think it might be a good idea to include this code that @Jailander produced as part of #35 into my tmux-based launch system (see #28 and https://github.com/LCAS/spqrel_tools/blob/master/tmux/README.md). I would then create a "config" section into https://github.com/LCAS/spqrel_tools/blob/master/tmux/spqrel-pepper-config.json and work through with @Jailander code?

marc-hanheide, 2017-07-20:
@jailander as I already said in #35 I think something is very wrong with that PR, so I close it. Could you tidy this up to make sure we really only get the changes you want us to have?

marc-hanheide, 2017-07-20:
Should now be done properly in PR #38

marc-hanheide, 2017-07-21:
First Implementation is merged. Needs testing

marc-hanheide, 2017-07-25:
not sure we use it right now, but it worked ;-)

closed 2017-07-25 by marc-hanheide