LCAS/spqrel_tools #9: Packing List

marc-hanheide, 2017-07-13:
- [ ] Japanese plug adapters (BOTH)
- [ ] Router + switch + eth cables (LINCOLN)
- [ ] Wired mouses (BOTH)
- [ ] UK extension lead (LINCOLN)
- [ ] ASUS Xtion

iocchi, 2017-07-18:
- [ ] Consent forms signed (DIAG)
- [ ] Consent forms signed (LINCOLN)

marc-hanheide, 2017-07-21:
I'll check in a very large suitcase, which should be big enough for our kit. Plan is that I pack it all Saturday morning when I have the car and pick up everything from the lab, ok @sergimolina ? Does anyone know if laptops are allowed in checked luggage?

iocchi, 2017-07-23:
laptop in checked luggage: I would say yes 
- any idea about Hokuyo Laser in cabin baggage?

marc-hanheide, 2017-07-23:
I think it should be fine. If you have it, bring a receipt or something, showing it's a standard device.

marc-hanheide, 2017-07-23:
I only have one Japan-UK Power Adapter, so if @sergimolina has any, bring them.

marc-hanheide, 2018-06-08:
Can the Lincoln team please pile all things to take in one place in the FabLab?

mtlazaro, 2018-06-08:
I printed consent forms for all, we can signed them there.

marc-hanheide, 2018-06-11:
@MFernandezCarmona @sergimolina @francescodelduchetto I put some bits from the list in the FabLab.

closed 2020-09-16 by marc-hanheide