LCAS/spqrel_tools #31: Initialization of Sapienza Pepper

iocchi, 2017-07-14:
In running 
`git clone --depth 1 --recursive -b pepper git://github.com/lcas/spqrel_launch.git`

we get the following
```
Cloning into 'spqrel_launch'...
warning: templates not found /install-git/share/git-core/templates
remote: Counting objects: 275, done.
remote: Compressing objects: 100% (208/208), done.
remote: Total 275 (delta 22), reused 265 (delta 21), pack-reused 0
Receiving objects: 100% (275/275), 13.16 MiB | 2.86 MiB/s, done.
Resolving deltas: 100% (22/22), done.
Checking out files: 100% (346/346), done.
fatal: Uh oh. Your system reports no Git commands at all.

```
The `worktree` folder contains only the names of the submodules, but nothin in
and git pull into these sub-folders does not update anything.

Should we clone the git repositories in worktree manually?

marc-hanheide, 2017-07-20:
Did you try the new procedure as documented properly in https://github.com/LCAS/spqrel_tools/wiki

I worked through it from scratch and this should now all be fine.

marc-hanheide, 2017-07-21:
Marc does it in Nagoya

closed 2017-07-21 by marc-hanheide