LCAS/spqrel_tools #32: Flex error on 14.04

iocchi, 2017-07-14:
The problem is that naoqi distribution does not include flex.
So I manually added file `FlexLexer.h` in `PNPnaoqi\include` (16.04 version)
You can try to replace this file with the 14.04 version (`/usr/include/FlexLexer.h`)
Maybe we can try in transforming this file in a symbolic link.

marc-hanheide, 2017-07-15:
Yes, it works with a symlink. But why is it needed in the first place, we cross-compile on a system where the headers are available, so, why do they have to be copied in?

iocchi, 2017-07-15:
I agree that it is not needed (it was a quick fix that I even forgot about...) probably we just need to add the include dir in CMakeLists file

marc-hanheide, 2017-07-15:
Indeed. Trying it now.

marc-hanheide, 2017-07-15:
see https://github.com/iocchi/PetriNetPlans/pull/20 works!

closed 2017-07-15 by iocchi