LCAS/spqrel_tools #37: Questions to TC during team leaders meeting

mtlazaro, 2017-07-20:
- Which is the variability in the formulation of the questions of SPR test

marc-hanheide, 2017-07-25:
Notes from Meeting 1 (6pm day 1):

* 50% to be promoted to stage 2, 4 teams in SSPL, the TC might increase the number if scores are close.
* final with two teams. One winner in each sub league.
* SSPL will not have a 3rd place as they are not enough
* all finals will be in ONE arena together

marc-hanheide, 2017-07-27:
# Meeting 3:

# General suggestions for TC

* in the future, allow contingencies, where the professional actor is asked to perform a specific action to allow to continue. After every contingency is applied, all points awarded afterwards to be halved, also for every following contingency again.
* provide food!

closed 2017-08-04 by marc-hanheide