LCAS/spqrel_tools #39: robot status UI

marc-hanheide, 2017-07-21:
to display:

- [ ] current plan
- [ ] current action
- [ ] temperature
- [ ] battery

iocchi, 2017-07-21:
current action stored in memory with key `PNP/CurrentAction`
current plan in `PNP/CurrentPlan`

marc-hanheide, 2017-07-21:
Are you raising events for those? Polling is not really efficient and also not following the event-driven pattern I use throughout ;-)

marc-hanheide, 2017-07-21:
I can't see this yet, did you commit and push?

marc-hanheide, 2017-07-22:
works! Thanks, @iocchi 

Only comment: The `PNP/CurrentAction` isn't emptied at the end of a plan. It would be better if it would raise a `""` content event at the end of an action, but that's not critical.

marc-hanheide, 2017-07-22:
Current status:

[image]

marc-hanheide, 2017-07-25:
@iocchi I don't know the key for Temperature. What should I display? The rest is working fine.

iocchi, 2017-07-25:
If you use the advanced menu on Pepper, you can read the name of the memory keys for the temperatures.
They are probably also somewhere in the documentation, but I think reading them from Pepper is faster.

iocchi, 2017-07-25:
Can we start a ful test from GUI only? Suppose network is fully unavailable, can we just boot the robot and use the UI to start a test?
How is logging activated? Can we have a button on the GUI or an action to enable/disable logging from the plan?

closed 2017-08-04 by marc-hanheide