LCAS/spqrel_tools #44: topological navigation issues

marc-hanheide, 2017-07-25:
- [ ] we need to facilitate other actions (for specific door pass action) The basics are there but need testing
- [ ] synchronisation problem, sometimes NAOqiPlanner manages to reach intermediate waypoint but TopNav does not carry on, the interaction protocol between NAOqiPlanner and TopNav seems brittle, @mtlazaro to indicate clearly on `NAOQiPlanner/Status` when the metric planner gives up, status `FAILED` and `SUCCESS`, The reason should be given in another ALMemory slot
- [ ] we need a recovery behaviour in `monitored_navigation`, simply "wiggle and retry *n* time" should be tried.

I ping @Jailander as he might have some input.

marc-hanheide, 2017-07-25:
@Jailander can you shed some light, or maybe also @pet1330:

Looking at https://github.com/LCAS/spqrel_navigation/blob/80b2d8e364664a399a4be4b3175138551a0ded1d/src/topological_navigation/scripts/topological_navigation.py#L237-L274 I don't understand what the if statements do and how they make sense.

```
        if self.current_node == 'none' and a not in self.move_base_actions:
            if a not in self.move_base_actions:
```

for me that is a tautology, if the first if was fulfilled, so is the second, or do I miss anything?

Then it continues with

```
        else:
            if a not in self.move_base_actions:
```

So this is called if `a` is not a move-base action, but only if `self.current_node` is not `None`? I'm not sure I understand all this. It could use some documentation ;-)

Jailander, 2017-07-26:
Hi, that is probably my mistake using legacy code from strands sorry, the logic behind that is that if the robot has to perform a non move base type action it must navigate to the waypoint and then perform the action. 

But if the action is move base type it can navigate directly to the next node in the list.

marc-hanheide, 2017-07-26:
Seems to be working now, will reopen if not

closed 2017-07-26 by marc-hanheide