LCAS/spqrel_tools #48: Overall plan

marc-hanheide, 2018-04-24:
# Abilities:

- Navigation (@mtlazaro)
  - using RGBD to create laser scanner
- Dialogue (@francescodelduchetto with remote input from @andreavanzo )
- People Perception (@sergimolina, with remote input from @robpinillos ) 
  - detection and tracking
  - identification / maybe demographics estimation
  - sound localisation
- Object Perception (@MFernandezCarmona )
- UI (@marc-hanheide)
- System integration and deployment (@marc-hanheide)

# Tasks

See our old document at https://docs.google.com/document/d/1hbKG8kDjDA8ulWavLRsRMJMjw0T4TgYOE_0dg1rUWzw/edit

MFernandezCarmona, 2018-05-10:
Hi,
About object detection. I managed to make Yolo use python and connect to nao and even recognize some objects/persons. Out of the box it works , more or less.
I had to do some tricks, so I forked the original one to this:
https://github.com/MFernandezCarmona/darknet

Next steps would be retraining the network for our own classes, but that is way out of the capabilities of my graphics card. Suggestions?

marc-hanheide, 2018-05-10:
Two options:

* Ask @maxhuettenrauch or someone else of @GeriNeumann s team if we could run training on of the GPU beasts we have (maybe @GeriNeumann has a better idea than darknet for out of the box object recognition anyway).
* Put the training job on either GPU enabled AWS EC2 or GCP computing. I'm happy to spend some money on that. I have created an AWS Account for L-CAS some time ago. It would cost around £1 per hour on a 2 Titan/8 CPU machine if I recall correctly.

soumenms2015, 2018-11-10:
does it work on windows platform?

MFernandezCarmona, 2018-11-12:
@soumenms2015 I'm afraid not. Please check Readme.md